frankiearzu/DSMTools #2: Spektrum Telemetry

pierrotm777, 2023-03-27:
Thanks franki for your new firmware.

I can see now the gps location on only one line but it return me bad value.
For :
GPS Latitude = 89.1234560 degree
GPS Longitude = 178.1234560 degree
I obtain:
 891°14'4.4''N et 1781°14'4.4''E

Did you have done the changes asked by mstrens ?
If yes, where are the source with these change please ?
May be [here](https://github.com/frankiearzu/edgetx/blob/v2.8.1_TextGen/radio/src/telemetry/spektrum.cpp) ?

Another thing, i discover always ID from 1800 to 180D
I think it's why 0x18 data are omited.
Spektrum code specify:
//////////////////////////////////////////////////////////////////////////////
//
//                      RX CAPACITY METER
//
//////////////////////////////////////////////////////////////////////////////
//
typedef struct
{
    uint8_t     identifier;                                                     // Source device = 0x18
    uint8_t     sID;                                                            // Secondary ID
    int16_t     current_A;                                                      // Instantaneous current, 0.01A (0-328.7A)      7FFF-> no data
    uint16_t        chargeUsed_A;                                                   // Integrated mAh used, 0.1mAh (0-3276.6mAh)
    uint16_t        volts_A;                                                        // Volts, 0.01V increments (0-16.00V)
    int16_t     current_B;                                                      // Instantaneous current, 0.01A (0-328.7A)      7FFF-> no data/sensor B
    uint16_t        chargeUsed_B;                                                   // Integrated mAh used, 0.1mAh (0-3276.6mAh)
    uint16_t        volts_B;                                                        // Volts, 0.01V increments (0-16.00V)
    uint8_t     alerts,                                     // Bit mapped alert conditions (see below)
                highCharge;                                 // High nybble is extra bits for chargeUsed_B, Low is for chargeUsed_A
} __attribute__((packed)) STRU_TELE_RX_MAH;

Thanks for your work :-)

Cordially,

Pierre

pierrotm777, 2023-03-27:
We see also an error with the GAlt value that stay always -1000m.

Thanks for your help,

Pierre

frankiearzu, 2023-03-27:
The explanation and source are in the original ticket:
https://github.com/EdgeTX/edgetx/issues/3368
Lets continue there...

closed 2023-04-09 by frankiearzu