frankiearzu/DSMTools #3: "Attitude Trim" menu always triggering [Level model and capture attitude] event

silicongarage, 2023-03-31:
When entering "Attitude Trim" menu a [Level model and capture attitude] event is always triggered with the touch screen [Back] button.  This means you cannot adjust the Roll and Pitch angles manually or enter the menu without disturbing stored trim values.
A simple example is that of configuring a tail wheel airplane.  Raising the tail to level and pressing "capture" will correctly pick up trim values.  However upon lowering tail back down and pressing [Back] will then incorrectly capture the current (and incorrect) attitude trims.

I'm still learning how the DSM TX<->RX communication works and how your code handles the complexities.  But it appears to me that it is somehow related to the first menu line definition of [Level model and capture attitude] and upon exit of "Attitude Trim" that then getting executed as a default selection. 

DsmFwPrgLib.lua
line 1559
```lua
Text[0x00CD] = "Level model and capture attitude/m"; -- Different from List_Text , and force it to be a menu button
```
As a work around solution, after capturing the correct attitude trims or manually adjusting trim values press the physical RTN button to exit the menu (and DSM fw prg) instead of the touch [Back] button.

frankiearzu, 2023-04-04:
I think i found the probem.    The "Back" needs a special handling to save the active values.

When navigating menus (also back, next), the current menu line needs to be sent to the RX.
in the base of BACK.NEXT,PREV i was sending 0.. looks like i need to send 80,81, and 82 instead.

DsmFwPrgLib.lua: 
Line 570.    dsmLib.GotoMenu(menu.BackId,0) needs to be dsmLib.GotoMenu(menu.BackId,0x80)..
Next is 0x82, and Prev is 0x81.

Can you try this changes and let me know if that fixed your problem??

[image]

silicongarage, 2023-04-06:
Thanks!
I assume you meant 0x80 (not 080) on line 570 in your code snapshot.
With these 3 changes the Trim values are now maintained correctly on entry and exit with [BACK]

frankiearzu, 2023-04-07:
Great! Thanks for testing it..  Took the screenshot before i saw my error!.  Will add it to the next version.  Have been working on a new version for older/smaller radios (smaller screen, and significant less memory).. what a pain dealing with low memory!  TX12. Zorro, etc.

frankiearzu, 2023-04-09:
Fixed. Released on Version 0.54

closed 2023-04-09 by frankiearzu