frankiearzu/DSMTools #5: Typo fixes and my personal preferences..

txkflier, 2023-04-17:
When I installed 0.53, I noticed several typos in the code and proceeded to fix them. I've now installed 0.54 and have corrected those same typos along with a few other things. Since I only use the color/touch version, it is the only one I've looked at. I probably made some changes to the three Lua scripts under DSMLIB as well. I'm sure you know how to diff my files against yours. I would have sent a diff file, but I don't have much experience with diff.

Since I changed "traileron" to "taileron", I renamed "tt_traileron.png" to "tt_taileron.png".

[DSM_FP_054_Edited_by_TxkFlier.zip](https://github.com/frankiearzu/DSMTools/files/11245554/DSM_FP_054_Edited_by_TxkFlier.zip)

Thanks for making this code available!

frankiearzu, 2023-05-28:
Fixed for v55

txkflier, 2023-05-28:
I’ll check it out..

closed 2023-05-28 by txkflier

ror1948, 2023-05-28:
where is 0.55 color?
I find it very unclear where to download the files

frankiearzu, 2023-05-28:
In development... will update the zip file later tonight.. not at home.