frankiearzu/DSMTools #6: Relearn Servo Settings  sets receiver frame rate to 11ms (problem??)

silicongarage, 2023-04-22:
edgetx: 2.8.1-TextGen-selfbuild
multi: 1.3.3.20 AETR
dsmtools: 0.54
receiver: 2.38.05

Not sure if this is a bug (as I see similar behavior on my NX10),  actual desired behavior or even something you could control with the DSMTools code.  

On the TX16 side I have set Internal RF -> Servo update rate to 22ms.
I would expect under RX Forward Programming that Relearn Servo Settings would respect this value but instead it populates the receiver frame rate channels to 11ms.

As noted, this is the same behavior I see on my NX10 with the frame rate set on the transmitter to 22ms, with the difference being there are independent fields to set (ha, and ignore) the frame rate on each channel.  When doing Relearn Servo Settings with NX10 all  set to 22ms the RX channel rates are changed to 11ms.

Should DSMTools/EdgeTX do (what appears to be) the right thing and respect the TX channel frame rate  settings when doing Relearn Servo Settings? (if that's even possible)  What would be the logic behind ignoring those (TX) settings and forcing receiver channel frame rates to 11ms?  This maybe a case of just duplicating existing DSM TX/RX behavior.

If DSMTools can (and decides to) respect the TX Model frame rate settings then you might also consider allowing configuration independently of each channel update rate as the NX10 does.

frankiearzu, 2023-04-23:
The "relearn servo settings" currently exchange  the folowing TX info (The use of each connection port: Thr/Ail/Elv/Rud and slaves to those), Servo Travel (-100..+100), SubTrims (not completely reversed engineered yet).
I haven't seen any message related to Frame Rate, but there is one message that currently don't know what it does.. maybe is for this.

For reverse engineering, i have to systematically change parameters in the plane on a real Spektrun TX, capture the messages exchanged between the TX->RX. and see what messages changed content.  There was one message that reagardless of what i changed, it was the same..   but never tried the frame rate... maybe thats what it is..   Let me do some experimentation and message captures to see if that message could be the missing behaviour.

frankiearzu, 2023-04-23:
Could not find in my Spektrum radio (DX6e) where do setup the servo frame rate. At Binding time, it always gives me 22ms.
Will look more.

silicongarage, 2023-04-23:
My NX10 is running 3.08
It's under System Setup ->Frame Rate
I forgot (hadn't used in a while) that the only option is to set Servo Mode to "Default 22ms" or  "Hybrid 11/22ms" .  Each channel is not independently configurable.

[image]
[image]
[image]

silicongarage, 2023-04-23:
Strange that NX10 doesn't even fully match "Hybrid 11/22ms" when Relearning servos THR channel also gets set to 11ms.

[image]

frankiearzu, 2023-04-23:
Don't have that menu in my Spektrum DX6e.. and no new firmware for it.   The radio seems to negociate with the RX the frame rate at bind time.

silicongarage, 2023-04-23:
Does it end up setting the receiver servo frame rates to 22ms?