frankiearzu/DSMTools #7: unable to select FM channel higher than ch6 on AR637

ror1948, 2023-05-27:
version 0.54 on a TX16s and Rx is a AR637T
Unable to select any channel higher than ch6 as FM CHANNEL for mode control.
I believe the AR637T has 9 channels but it is best to NOT check the range of channel entered.
Leave that up to the programmer.

qsiguy, 2023-05-27:
I believe to use channel 7+ have your switch programmed for the channel and rather selecting the channel in forward programming just click it and then flip the switch to auto select.  Then it should retain the correct channel.

qsiguy, 2023-05-28:
I just posted a video on my channel how to do this.

https://youtu.be/jLRE6MVU2hU

frankiearzu, 2023-05-28:
Is documented that you need to use the switches in this case,

closed 2023-05-28 by frankiearzu