frankiearzu/DSMTools #8: Bad Argument #1 Error

checkmatehau, 2023-08-08:
When trying to use DSM Forward Programming (Color+Touch) on my Radiomaster TX16s with EdgeTX, I get the following script syntax error popup:

`bad argument #1 to 'write' (FILE* expected, got nil)`

I am assuming this has to do with the file structure. How do I resolve this issue?

qsiguy, 2023-08-08:
What were you attempting to do when you received this message?  If during the model setup, which isn't actually in forward programming, perhaps you don't have the "/MODELS/DSMDATA" folder created?  When you do the model setup it writes a .txt file to that folder and if the folder does not exist you'd get an error.

checkmatehau, 2023-08-08:
I am an idiot and did not install ALL of the folders. I thought just the ones for Color+Touch were required. Closed.

closed 2023-08-08 by checkmatehau

frankiearzu, 2023-08-08:
Thank you guys for helpin!