frankiearzu/DSMTools #9: Avian ESC Forward Programming w/ Radiomaster TX16s

checkmatehau, 2023-08-08:
I am trying to add thrust reversing to a Habu SS 50mm EDF jet and this seems to be the only software to do that with a Radiomaster TX16s (EdgeTX), even though it seems to be tailored to helis. 

This jet has an AR631 Rx and an Avian ESC. I know this software is for AR636 Rx's, but when trying the TextGen, the screen says there is a Smart ESC and gives the steps for ESC programming similar to Spektrum Tx's with Avian Programming (i.e., Enter Menu, Step 1, Hold 5-10s, Low Throt, etc.). The issue is that I can't continue in this menu. After placing the sticks inward (like in the tutorial) it never goes to the next step. Is there a reason for this? Am I totally off-base on the purpose of this program?

This is sort of a "hail mary" on my part. I tried using an Avian Programming box from Spektrum and even THAT didn't work. I figured this method would at least confirm if I can add reverse thrust braking or not.

qsiguy, 2023-08-08:
The "DSM_AR636_Tel.lua" is the one for Heli's.  The one that contains the TextGen app used for the Avian ESC programming is DSM_SmartRX_Tel.lua

Make sure you have full rates set for the stick movement, it needs to see the values at full -100/+100 (or close to it) when you move the sticks.  If you have low rates on the weights aren't high enough to register by the Avian.

Second make sure you do the stick movement within 30 seconds.  Here is a link to a recent video I posted on setting up reverse using this app with an Avian Lite ESC and an AR631 on an AeroScout. Should be exact same process.
https://youtu.be/C0L_cKdJAy4

checkmatehau, 2023-08-08:
You're a legend. All I was missing was the 30-second timing this whole time... The amount of time and effort the past few days... Thank you, good sir.

frankiearzu, 2023-08-08:
Thank you guys for helping!