frankiearzu/DSMTools #11: F-mode only shows 2 and 3

ror1948, 2023-08-16:
edgetx 2.8.5 date 2023-08-04 time 09;49;14
did all the setup but my ch5 fmode only shows FM2 and FM3 in the FP setup
once I saw it display 191 instead of 1 and when I switched ch5 once all I saw was 2 or 3
2 was showing in pos 1 as well as pos 2
my ch5 is going from -100 to +100

i tried 0.54 all releases
also tried 0.55 prerelease

ror1948, 2023-08-16:
TX16s

ror1948, 2023-08-16:
Rx is AR637t with latest firmware

frankiearzu, 2023-08-16:
Please check that "Enable Max Throw" if OFF in the module settings (where you do bind).    If is ON, it will shift the servo range and you will never get Flight mode 1

txkflier, 2023-08-16:
I wish the EdgeTX developers would make their OS show the true pulse width in Spektrum mode.

frankiearzu, 2023-09-01:
Not a script problem

closed 2023-09-01 by frankiearzu