frankiearzu/DSMTools #12: FM Select channel not recognized

eljaydub, 2023-09-12:
I'm having an issue with Flight Mode channel selection. I'm using the AR8360T 8CH receiver with the Radiomaster Tx16S MkII, EdgeTX 2.9 and DSM Forward Prog 0.55. I've got the receiver in a Bush Mule V2 so all 8 channels are being used for servo or ESC controls. On page 15 of the [AR8360T manual](https://spektrumrc.com/ProdInfo/Files/SPMAR8360T-Manual-EN.pdf) under Channel Limitations, it says that 7 additional channels are available for AS3X and SAFE functions. I've got the three position switch SE set to Channel 9 for Flight Mode selection and momentary switch SH assigned to Channel 10 for Panic mode. 
The problem I have is that when I go to F-Mode settings and try to use Channel 9 as my FM Channel it does not work - switch SE does nothing and I am stuck in FM2 (corresponding to an input value of 0). If I change FM Channel to Ch5 (where I have my flaps) or CH6 (where I have my cargo door) the FM channel selection works just fine; I can switch between FM1, 2 and 3 no problem. Similarly for the Panic channel, the value appears to stay 0 if I select any channel number greater than 8. 
It seems that while the receiver manual says that it allows any channel up to 20 for AS3X/SAFE functions and that CH5 and CH6 do not need to be used for those purposes, there is something in the software somewhere that is not allowing any channel number greater than 8 from being assigned to AS3X/SAFE functions.

frankiearzu, 2023-09-12:
The RX itself is 20 channels, but the multi-module has a limit of 12 (don't know why).. left the script to only use those 12.

I think your radio is not transmiting channels higher than 8. 
In the Model-Settings -> Internal RF (where you do bind), make sure that "Enable max. throw" is OFF, and that the channel range say Ch1-Ch12  (Default is Ch8),

eljaydub, 2023-09-12:
That was indeed the problem! Thank you very much. I've run through all of the settings now and it looks like everything is working as expected. Next is the most exciting part - the flight test!

frankiearzu, 2023-09-13:
Not a bug!

closed 2023-09-13 by frankiearzu