frankiearzu/DSMTools #14: RX Connection Drop error

lienmeat, 2023-09-29:
With v0.55a, when trying to go to Forward Programming RX, it has only one button "Gyro Settings".
When I click on "Gyro Settings" after a second or two of an empty screen a dialog box pops up that says:
"Script syntax error
RX Connection Drop"

I don't think my RX connection did drop, because I can still control the plane after that.
Happens every time for me, can't get past it.

This is with an AR631 v2.38.4 if that matters.

qsiguy, 2023-09-29:
That means you have a locked receiver. In order to access it, even with a Spektrum transmitter, you must update the receiver firmware. I demonstrate this in episode 2 of my video series on this script as well as how to unlock it and reset it.

This is not a bug or limitation of the v0.55 script or EdgeTX. 

https://youtu.be/bzWeMaahJhk?si=VQFNhNdpyu370osS

frankiearzu, 2023-09-29:
You are using EdgeTX or OpenTX??
The connection drop means that the RX did not responded to forward programming commands sent from the TX.. it tried 3 times and if no answer, it gives up and report the error.

If you only get "Gyro Settings", that means that your receiver is locked (from a BNF plane).  What model is it??

Most BNFs will have at leat "Gyro Settings" and after "SAFE Select". but some BNFs, even restrict the "SAFE Select".

In your radio, after you duplicate the error, there should be a log file in the /LOGS/dsm_log.txt..   can you attach it please.
That Log has the conversation between the RX and TX and we can see what the RX send (or did not send).

For that RX, the latest version is 2.46.1, where they allow a bit more control to at least been able to unlock it without using the the USB cable... but most don't have the cable that is used to unlock or update firmware. If you have the USB cable, try to upgrade the firmware.

closed 2023-11-07 by frankiearzu