frankiearzu/DSMTools #15: Avian ESC programming with X-Lite en MPM module not working with TextGen

RealTadango, 2023-11-01:
Hi,

I have a E-Flite D18 BNF and i use a FrSky X-Lite with the latest Edge TX (2.9.1). I have created a new model and did the binding. All control movements are correct (Reversed Aileron and Rudder). I can see the battery voltage using the scripts and can change the 631 gyro settings ect. I also see data from the Esc status menu (pack voltage). 

Now i try to get the ESC to activate reverse but TextGen is not showing anything when i select it. Is that normal? I don't see the stick command instructions as i expected. If i do the movements within 30 sec of powerup nothing shows up either.

RealTadango, 2023-11-01:
Btw, i used [DSM_Telemetry_v11_EdgeTX_Aug10.zip] and then the DSM_SmartTX_Tel.lua

RealTadango, 2023-11-01:
I just found out this new esc does not support TextGen yet :( Thanks for nothing Spektrum.....

closed 2023-11-01 by RealTadango

frankiearzu, 2023-11-01:
What ESC are you using??  I know that there a some who cannot do TextGen (Locked), is it an older version?

RealTadango, 2023-11-01:
It is a SPMXAE0240A. A Lite Dual 40A 3-4S esc. What i can find online is that these duals don't have TextGen in the firmware yet.

txkflier, 2023-11-01:
The same ESC came in my E-Flite Beech D-18 and I couldn’t program it from my Spektrum iX14 transmitter.

qsiguy, 2023-11-01:
When you are bound to the receiver/airplane and open the TextGen script, do you see the menu?  If so, you are connecting to it.  If it isn't connecting at all you won't even see the TextGen menu appear as it is populated by the ESC via the RX.  I have successfully programmed the SPMXAE30B which is the Avian Lite 30A ESC.  I would be surprised if that smart dual channel ESC wasn't TextGen compatible.

RealTadango, 2023-11-01:
It does not show anything. This a limitation of the dual escs as I understood. I have ordered the LCD box and will check if they have 2 CPUs and have different settings. Maybe that is the reason......

qsiguy, 2023-11-01:
There are two versions of this dual ESC, the #SPMXAE0240A and the #SPMXAE0240B.  The 'B' comes in the Twin Timber and I know for a fact that it can be programmed via TextGen.  The D18 comes with the SPMXAE0240A so apparently the 'A' model can only be setup with the programmer.  Wonder if they have a firmware update yet that can be done for it?  Please update this post once you try the programmer.

FYI the single yellow wire on the ESC plugs into the port labeled "ESC" on the programmer on the "S" pin and the other ESC connector goes to the programmer "BATT" connector. Manuals for these things from Spektrum is not very good and in many cases non-existent.

RealTadango, 2023-11-02:
I just tried the Program box (With both program leads) and as expected there are 2 separate ESC's inside. Both have their own settings. If you are programming ESC 1 then ESC 2 will still work normally so there is not a single central CPU that is controlling both. I enabled reverse and it works. I will check later if there is an firmware update for it.