frankiearzu/DSMTools #16: DSM Failsafe forward programming not working with AR6610t receiver

BergWolfRC, 2023-11-15:
Hi Frankie,

first of all thank you very much for providing these great DSM tools. I was really glad I found them.

I tried them out today in the following setup to do failsafe position forward programming:

- Using you DSM Forward programming version 0.55a multicolor mode
- TX is Radiomaster TX 16 S MKII
- 4 in 1 module is used
- operating system is EdgeTX 2.9.1
- Receiver is a Spektrum AR6610T (6 channel telemetry receiver with built in altitude and variometer)
- receiver is properly bound to TX and more than 8 channels are configured (running a 4 servo wing v tail glider of the template from Mike Shellim (https://rc-soar.com/opentx/setups/f3j/index.htm)
- all servos operating properly
- telemetry data is returned to the TX

These are the steps that I performed to do the failsafe forward programming:

- powered on receiver and transmitter and verified model reacts to stick inputs
- launched "DSM Forward Prog v0.55a (Color)"
- ok'ed the info message that appeared
- (did not do any model setup as all the programming is TX side and no save or smart or AS3X required)
- selected "Forward programming RX"
- the progress messages of "sending channel 1-12 appeared 
- orange box "settings" appeared
- clicked on "settings"
- Clicked on "Failsafe" in the next panel
- Put sticks into failsafe position
- selected "Capture Failsafe Positions"
- selected "Apply"
- message "Complete" appeared
- exited the DSM forward programming tool

Now, when powering off the transmitter I would assume the receiver to tell the servos to go into the failsafe positons as commanded by my stick input when I captured the Failsafe positions. However, the receiver just holds the servo positions as the were when I powererd off the TX.

Please tell me what I did wrong or which additional info you need from me to debug and fix.

Thank you very much

 Wolfgang

frankiearzu, 2023-11-15:
Check this video, it explains how failsafe works.. Also the Failsafe behavior is affected if the RX has SAFE capabilities.  
(i think the AR6610t  don't have SAFE, don't have that RX but test it with a AR631)
https://www.youtube.com/watch?v=2lzXlROWaJw

Remember that the tool is only displaying the information sent by the RX.. all the logic of what functions to do resides inside the RX.

When it NOT configured to go into a SAFE flight mode: 
On the Failsafe screen, only the ones setup as "Preset" will be captured.
The positions will be captured when you press "Apply", not when you entered the "Capture Failsafe Menus".
After that, you can go back to "Failsafe" and check the values captured for every channel.

When IS configured to go into a SAFE:  
The plane will be in SAFE Auto-level and the attitude is configured in "FailSafe Angles".
This is all in "Gyro Setting->System Setup->SAFE and panic mode setup
With this, you can setup the plane to fly itsels in circles  when in FailSafe.

frankiearzu, 2023-11-15:
By the way, I have been fixing SoarETX to work with F3J/F5J planes.
You can check the code here.  The main difference is that all the setup of the wings and mixers is done from the GUI.
Also the main screen captures your flying time and in F5J, your motor time/altitude.

https://github.com/frankiearzu/DSMTools/tree/main/Planes%20and%20Templates

BergWolfRC, 2023-11-16:
Thanks for all the info.
I ensured the sticks are in the desired failsafe position when I hit the Apply button.
Still no failsafe positions captured.

In contrast from my DX-9 I can perfectly set the failsafe positions via forward programming for that very same receiver.

What happens, i.e. which code sections are executed and which data is sent to the receiver when I press Apply?
Similarly what happens between the Apply and the "Complete" message?
Did I understand you right that not only should it display "Complete" but also it should display the captured ouput positions?

Thanks
 Wolfgang

BergWolfRC, 2023-11-16:
Solved:

I did various test and in the end the problem was that the channels that I used for testing were for some reason set individually to "hold" as opposed to "preset". After setting the to "preset" I could successfully capture any desired failsafe setting.
Thanks for your help.
Feel free to close this issue which turend out to be a typical pebcak :-) "Problem Exists Between Chair And Keyboard".

I will try out the new open soar templates ASAP. 

Thanks
 Wolfgang

BergWolfRC, 2023-11-16:
Looked at the new SoarETX templates today. What you guys have done there is quite cool a spearation between the parameters and the wiring together of everything. This is exactly what is needed to prevent users from accidentally screwing up the whole model setup.
I alse tested the SoarETX v2 templates today.
Again on my TX16s MK II with EdgeTX 2.9.1. 
Started with  F3J_v2. 
Unfortuantely none of the menus on page 2 are selectable and clickable (Outputs, Switches, Wing alignment, Airbrake curves, Mixes & Battery, Graph).
How can I enable them? 
And is there a way to get those scripts also into the EdgeTX Companion Simulator?

frankiearzu, 2023-11-16:
They are widgets..  to "enable" them, press and hold, until you get the menu, and do "full screen"

BergWolfRC, 2023-11-16:
That did the trick.
Thanks
Is there any docu about those templates available that would explain which globvars server for what etc.

closed 2023-11-18 by frankiearzu