frankiearzu/DSMTools #17: DSM_FP_056_COLOR_Dec5..zip Model Setup Gives LUA Error

maccers, 2023-12-08:
When running the DSM Forward Programming on my TX16s I get a LUA error when running **Model Setup** and **Wing & Tail Channels**. The setup works on version v0.55a, not v0.56. 

TX16 Version: 2.9.2, using [tx16s-edgetx-v2.9.0-ratio-txt.bin](https://github.com/frankiearzu/DSMTools/blob/main/EdgeTx_Firmware/tx16s-edgetx-v2.9.0-ratio-txt.bin)

frankiearzu, 2023-12-10:
Can you try the new one  dated "Dec9".   Found a problem when starting a brand new plane..  editing an existing one did work,

frankiearzu, 2023-12-10:
Also unless you are using "ratio" to adjust telemetry data values, use the official 2.9.2 firmware.  Will compile a new 2.9.2 + the ratio fix if you really needed

closed 2023-12-24 by frankiearzu