frankiearzu/DSMTools #18: [Request] Ethos compatibility

mszulen, 2023-12-10:
Is there a chance for Ethos compatibility of the tool?
Thanks!

frankiearzu, 2023-12-10:
Really don't know what it would take. don't have a FrSky radio.
Also there is a lot of plumbing in the EdgeTX firmware to support the Multi-Module and Forward Programming beside the LUA scripts.

mszulen, 2023-12-10:
Thanks for a quick answer. I wasn't aware of the patches. The multimodule itself is working with Ethos (Frsky have provided the basic LUA). Now, the situation is more clear to me. Thanks again!

Crunch69, 2024-05-04:
FrSky uses multi module and would also require forward programming.  It would be great to add this to ethos

Guy559, 2024-05-19:
This seems like a terrific tool. 2nd the desire to see it ported to include Ethos.

Crunch69, 2024-06-16:
Is DSM tools tightly integrated with multi module if it is would it be easier since multi module is already working with full Telemetry in ethos?

frankiearzu, 2024-08-21:
Will track it on duplicate ticker #21

closed 2024-08-21 by frankiearzu