frankiearzu/DSMTools #19: Not able to enter TextGen

steffku, 2024-05-28:
I have an E-Flite Night Timber X. It has a Smart receiver (AR637T) and an Avian 60A ESC. If i enter TextGen (newest version of DSM tools with EdgeTX 2.9.3, TX16S), there's no text displayed, I ran through the troubleshooting from provided link to the Spektrum youtube video without success. Is there anything else that i can check (maybe the software version of receiver or ESC is too old?!)?

frankiearzu, 2024-06-01:
TextGen uses the telemetry channel to send the MENU data back to the TX.
Make sure that the protocol is DSM X2F, and do discover sensors to make sure that you are getting back telemetry from the RX.  

The timber is not that old, so it should work with TextGen.

frankiearzu, 2024-06-02:
Just double checking, is the ESC connected on the first port (Ch1) of the RX, right?
Are you getting telemetry data from the ESC?  Sensor Erpm for example?

steffku, 2024-06-02:
I checked the following points:

- ESC on channel 1 of RX
- telemetry on and in state "discover new sensors"
- enter TextGen menu immediately after power on
- full travel on rudder and elevator
- protocol is DSM X2F

still no success. It's strange. I still cannot see the TextGen menu.

qsiguy, 2024-06-02:
You say it is "in state of discover new sensors" but when you click "discover new" does it actually populate the list? Don't bother going into the TextGen menu if you aren't seeing data in the radios telemetry list. Need to sort that out first.

qsiguy, 2024-06-02:
Also, are you certain it has the Avian ESC? My older Timber X (not the night Timber) does not have the Avian, make sure your Timber isn't older and pre-dates the Avian ESC.

frankiearzu, 2024-06-02:
@steffku   do you see anything at all in the TextGen??   Just the "TextGen" title, or are you getting the instructions of how to get into the Avian configuration (Like the instructions to move the stick to specific positions?).

As a @qsiguy said, check that is an Avian.. could be older ESC.

qsiguy, 2024-06-02:
Sorry one more, if you haven't come across this video of mine it may help you find something.
https://youtu.be/C0L_cKdJAy4?si=G8gVkEFPROo0MWxw

qsiguy, 2024-06-03:
I tried many scenarios on my AR631 and a 60A Avian ESC and with the protocol set to DSM X2F, ESC connected to channel 1 I cannot get it to NOT connect via TextGen and show text.   Make sure your ESC servo connected to channel 1 is turned the right way around so you aren't swapping the data and ground wires.

Do you have throttle control from the radio and everything else working and just cannot get TextGen to show text?  As I stated above, make sure you really have an Avian ESC.  Next, are you sure you have the DSM_SMART_RX_Tel script installed correctly and it's the latest version?  1.2 is the latest version although 1.1 should be fine as well on the TX16S.

Last thing I can think of is if the ESC is, in fact an Avian, perhaps it needs a firmware update?  I would not think it wouldn't connect to the TextGen app but could be.

steffku, 2024-06-03:
Hi guys,

thank you very much for your help. I appreciate that!

I'm sure that i have an Avian ESC installed. Afaik, all versions of the Night Timber X have a Smart RX and an Avian ESC installed. I bought this 2 years ago (May 2022). The manual also says, that there's an Avian ESC installed.

@frankiearzu I don't see anything in the TextGen menu. Just the word "TextGen" in the upper left corner.

@qsiguy I'm sure that i have connected the ESC on channel 1 the right way. Throttle works as expected. I also have the newest version of the DSM Smart RX Tel script (version 1.2).

Until now i tried EdgeTX 2.9.4 and 2.10.0. I will try 2.8.5. Which version do you use?

qsiguy, 2024-06-03:
I've been using the scripts since about 2.8.4 I think and used in each new release. Currently up to 2.10.1.  Can you post your model yml file for us to analyze? Perhaps we can identify something there?

When you try to connect to TextGen your receiver and ESC are both powered up and bound to the transmitter correct? If not you won't see anything because the menus are populated by the ESC.

steffku, 2024-06-03:
Yes, both are powered up. Everything works fine, except TextGen.

See the model file attached.
[model24.zip](https://github.com/user-attachments/files/15537362/model24.zip)

qsiguy, 2024-06-03:
I didn't see anything in the yml file that would effect the TextGen functionality.  If you leave off the SOUNDS and IMAGES folders to keep the size down you can zip the rest of the SD card and we can check that out, see if something with the lua is wrong/corrupt?

Any chance you have a friend with a different transmitter, perhaps a Spektrum model to see if that will connect.  If a Spektrum radio won't connect either that could rule out the radio and point towards a firmware issue with your ESC or RX.    Can you access Forward Programming to the RX?  If so, what is the RX firmware version?  It will display it on the screen once you are in Forward programming.

steffku, 2024-06-04:
If i enter Forward Programming, i get the error message: "No compatible DSM RX". So i assume, there's an issue with the RX firmware. Don't know, if there's a chance to update the firmware. Receiver is an AR637TA.

frankiearzu, 2024-06-04:
That point that the RX is not sending data back.  are you sure that your telemetry sensors works??

You should have a "FM" sensor changing every time you lip your Flight mode switch (different number),  and "ERpm" that is reported by the ESC as the RPM of the motor.

frankiearzu, 2024-06-04:
Can you check that you don't have Telemetry Disables in your bind page.
On newer versions of EdgeTX is no longer an option, but in older version, you can turn ON/OFF telemetry.  Don't remember in what version was that removed.

[image]

steffku, 2024-06-06:
Yes, i use 2.10.1 now:
[image]
And i have both sensors.
FM:
[image]
and Erpm:
[image]

frankiearzu, 2024-06-06:
Ok, so you have telemetry back from RX to TX..     
Then my only other theory is that your RX is locked, if you cannot get to Forward Programming or TextGen.   There is a new version of multi-module v1.3.4.x, but should work with the one you have.

Really ran out of ideas.. if someone in your club has a Spektrum radio, you can bind your RX to it and check if Forward prog and textgen (last page of telemetry screens) works.

txkflier, 2024-06-08:
My TX16S has EdgeTX 2.9.3. I had never tried TextGen before, so I just ran a test to see what would happen and it works fine for me. I can edit the ESC settings. I can't imagine what you're doing wrong. I didn't do anything special.

[image]
[image]
[image]
[image]

steffku, 2024-06-08:
> Ok, so you have telemetry back from RX to TX.. Then my only other theory is that your RX is locked, if you cannot get to Forward Programming or TextGen. There is a new version of multi-module [...]

Yes, i agree. I also assume that my RX is locked. Maybe i will purchase a programmer cable and try to unlock it. But don't know if it is worth the 30€.
So thank you all for your help. If i can manage to unlock my RX i will write here, if this was the root cause.

txkflier, 2024-06-08:
I just loaded my AR630 with firmware V2.37.8, reset it, and used FlightEngr's SASMB to set it up as a locked receiver. I could still access TextGen and read the ESC's settings. I could also go into Forward Programming, enable SAFE Select, set up the flight mode channel, and a few other things. It appeared to be locked to me, but everything was still working.

steffku, 2024-06-13:
Now, i'm one step further: i bought a programmer cable and updated the RX to the latest firmware. Now forward programming works but TextGen still doesn't. I assume the ESC also needs a firmware update. That means i have to purchase an ESC programmer for another 30€. Spektrum knows how to make money!

txkflier, 2024-06-13:
Yes, everything should be running the latest firmware - transmitter, receiver, ESC, Sky remote ID, etc. Otherwise, one component might not know anything about the others.

closed 2024-06-22 by frankiearzu