frankiearzu/DSMTools #20: Error writing to file when using forwardprogramming lua script

kootje1976, 2024-06-14:
[image]

we extracted the zip to the root folder as described:

SD_FP_56_COLOR_Dec9.zip

when starting up the plugin, we receive the error in the image

do you have any idea what can be wrong?

frankiearzu, 2024-06-14:
We are aware that sometimes, writing to Log files on EdgeTX 2.10.x is having problems.  Still trying to find a solution on EdgeTX.
https://github.com/EdgeTX/edgetx/issues/5077

In the meantime, use v0.57  (on the "in Development", but will move it to the main directories).   Is Exactly the same as v0.56, but with writing to the logs turned off.  The Logs are mostly useful to debug new version of RX firmware.. and with 0.57, we can turn it ON as needed, but the default if OFF.

frankiearzu, 2024-06-22:
Strange Problem with EdgeTX, not the LUA script. closing this one here.

closed 2024-06-22 by frankiearzu