frankiearzu/DSMTools #21: DSMTools can this be ported to work on frsky ethos

Crunch69, 2024-06-15:
[DSMTools](https://github.com/frankiearzu/DSMTools)
Public

frankiearzu, 2024-06-22:
Have been looking into that, but is not that easy.. only have access to the PC simulation, and really don't know if Ethos will have the internal plumbing to support Forward programming (what is called a Multi-Buffer in OTX/ETX lua).

mszulen, 2024-08-01:
> Have been looking into that, but is not that easy.. only have access to the PC simulation, and really don't know if Ethos will have the internal plumbing to support Forward programming (what is called a Multi-Buffer in OTX/ETX lua).

Perhaps, worth checking with the developer of Ethos? https://github.com/bsongis

bob01, 2024-08-13:
This would be fantastic!

frankiearzu, 2024-08-21:
Started to do the port....
LUA on ethos is missing the functions to communicate with the Multi-Module (using LUA), but opened a ticket for an improvement on their Github.

In the meantime, i am using an RX simulator to help develop faster the UI on the simulator.

Some preliminary screenshots: 
[image]
[image]

jbeard6, 2024-09-10:
I'm also subscribed and willing to test with a Blade Fusion 180 on a FrSky Tandem X18.

frankiearzu, 2024-10-31:
Good progress so far... FrSky did the modification to talk to the MultiModule on the 1.6.0-preRelease, and is working.
Now what im missing is getting some info about the model output channels (like name), but will be solved soon.
Have ported the telemetry Scripts (where TextGen lives), and Forward Programming is talking to the RX.
The UI still need improvements, but the basic operations are working.

[image]
[image]
[image]
[image]

frankiearzu, 2024-12-09:
More progress on this over the last 2 weekends of work.
Looking at Rob's RotorFlight Suite, gave me many ideas.

## DSM Tools Suite (Ethos Version)

NOTE:  Requires Ethos 1.6.0 or newer
A release candidate is posted and should work on Simulators (with fake RX data), as well as real radios.

### Main Suite Page
[image]
[image]

### Setup app
[image]
[image]

### Forward Prog
[image]
[image]

### Telemetry 
[image]
[image]
[image]
[image]

frankiearzu, 2025-01-11:
We are calling this a DONE and closing this ticket.
The version is "DSMTools Suite v2.0 Final" which includes the latest changes up to RC4, as well as message changes to support the new Spektrum AS3X+ v3.2.7 (Jan 7, 2025)

closed 2025-01-11 by frankiearzu