frankiearzu/DSMTools #22: FP does not work on Spektrum AR631 V2 Rx

ror1948, 2024-06-22:
limited menu on ar631 v2
not enough to program Rx.
when i try to enter forward programming option on Rx it says no DSM compatible Rx present.

frankiearzu, 2024-06-22:
Didn't know that there was a V2.
Verify the following:
1. That is using DSM-X2F protocol
2. Make sure that you are receiving telemetry data; do discover sensors and see that they are refreshing.

By any chance do you have the USB programmer? To verify the firmware version or upgrade it to the latest.

ror1948, 2024-06-22:
yes i have usb programmer and i have already updated to latest(2.48.the V2 shows up when i go into Spektrum programmerFP works ok on my updated(2024) DX9 Tx

I was not using X2F
i tried X2F and it seems to be working nowyes telemetry is there and updating.
thanks

frankiearzu, 2024-06-22:
Ron, did not understand your last message.  Did it work after binding using DSM-X2F??

ror1948, 2024-06-22:
yesthanks

closed 2024-06-22 by frankiearzu