frankiearzu/DSMTools #26: Possible bug when using a two-servo delta wing

JoeSa3rd, 2024-09-15:
I think this is a possible bug to report.
Setup
Jumper T18 - EdgeTX 2.10.4
DSMTool v0.59
Spektrum AR631 6-Channel AS3X/SAFE (firmware v2.48)
Two servo Delta wing (PopWing)
I have a small delta wing and have configured a two-servo delta wing mixer in EdgeTX. Test flown and trimmed the plane and then wanted to set up AS3X and Safe on the installed AR631. I’ve run through the model setup and forward programming first time setup & configuration for everything (AS3X & Safe) but what I’m finding is that AS3X works perfect but when I put the plane into Self-Level with AS3X mode it just rolls quickly and hard to the left. Even when I test Self-Level on the ground the control services don’t respond correctly. I’ve verified the configuration many times with no issues. Servo reversing, orientation of the RX, Etc. It’s as the RX does not understand a delta wing config, or the orientation of the RX is not correct, or some bug. Any thoughts or suggestions on what to troubleshoot?

Thanks,
Joe

frankiearzu, 2024-09-15:
Hi Joe,
Do you mind sending me a screnshot of your servo reverse page.. something like this:
[image]

Are you using the Elevon/Delta A or B???
The right hand side gives some info of the role of the channel.   **AilEle**  is to let it know that is is both Ail+Ele.  **M_Ele** is to tell wich one is the master, and **Slv** who is the slave.   Usually the difference between Elevon/Delta A or B is who is the slave.

JoeSa3rd, 2024-09-16:
Thank you... Your suggestion of setting of the config this worked perfectly and has fixed my issue in self-level mode -> "Deltas are a bit more complicated to setup since is 2 axis the gyro.. The A/B and reverse change the behavior.. but instead of just 2 options, now you have 4 things that can change. Focus on one axis at a time...Try first to make the Ail works well (reversing gyro as needed). Then focus on the Elevator using the A/B options."

closed 2024-09-16 by JoeSa3rd