frankiearzu/DSMTools #28: No compatible DSM RX…

glipski50, 2024-10-03:
TX16S, EdgeTX v2.10.4
DSM Forward Prog v0.59
AR637T v3.1.3
After starting Foreard Programming RX I get the message „No compatible RX…“ for 1 sec.
[image]

but then the channel infos are send normally
[image]

and I can continue programming
[image]

All programming options can be selected and looks like programming works without problems. Is that only a „cosmetic“ error or should I be careful?

Same behaviour I have with a AR610 with latest v2 firmware.

frankiearzu, 2024-10-03:
I think you have a mix of files in the DSMLIB.   Delete the /SCRIPTS/TOOLS/DSMLIB and copy the folder/directory from  zip file (or expanded zip file).
I just did a clean installation based on the zip file, and everything works.

glipski50, 2024-10-03:
Will do that first thing tomorrow morning (Europe).

glipski50, 2024-10-06:
Some more testing the last days. I deleted /SCRIPTS/TOOLS/DSMLIB and copied folder7directory. I than started DSM Forward Prog, did the model setup and started then FORWARD PROGRAMMING RX  At first everything went well.

During complete model setup with AS3X and SAVE parameters and setting the correct Gyro Channel Deflection in GYRO CHANNEL REVERSE I got sometimes, not every time when I started FORWARD PROGRAMMING RX, the error message agein „No Compatiblöe RX…“ and obviously the revceiver did not get updated, gyro channel direction did not change. 

When I got this error message I restarted transmitter and receiver and then it worked, but after starting FORWARD PROGRAMMING RX several times I got the error message agein. I tried to find a pattern but I could not see any. It just happens intermittantly.

frankiearzu, 2024-10-06:
Have the same versions, except not LBT.. but i think in DSM, there is no difference.

Does the empty lines got fixed? (the colons desapeared?), and are clean empty lines?     That is just to make sure that the firmware is correct:
[image]

The connect/disconnect could be related to the  quality of the communication link.   Check that you are using protocol DSM-X2F and that your RX is at least 2' (30-40cm) away from the TX.
On the telemetry page, check that the telemetry sensor values refreshes at least once per second.. and that nothing turns red. Red color is used when the sensors are not refresh fast enouth.

glipski50, 2024-10-06:
Screen looks exactly like yours in your second screenshot. Protocol is DSM-X2F, rx is 2+m away from the rx. Telemetry is refreshed normal, Alt and Vspd much faster than the other values. Firmware came from the source you mentioned. SD is a industrial one, tomorrow I will reformat it. My TX16S is a first edition, not a mk2.

The error messge comes up about 4 out of 10 times.

No clue, what I have to look at which I have not checked yet.

qsiguy, 2024-10-07:
What size is your SD card? Larger cards can have slower access times on the TX16S and could cause the script to run slower perhaps. Of course make sure it is formatted FAT32.

glipski50, 2024-10-07:
8 GB, FAT32, of course. 
Next try, I will change partition to 4 GB and format with block size 4 kB.

glipski50, 2024-10-07:
No luck.
Initialized and formated sd. Copied all files back to sd. Copied newly downloaded DSM Tool luas to sd. Same result as before. around 4 out of 10 times I egt the error message.

frankiearzu, 2024-10-07:
The SDCard should not be a problem.   Is related to the communication, thats why the connection drops (not receiving data more more than 2s).  
Can you send/attach the log after using the tool and runs into problems.. should b in /LOGS/dsm_log.txt

Also we can try to turn OFF the logs to see if that helps, the line with DEBUG_ON = 1 needs to be changed to DEBUG_ON = 0
[image]

Has to be related to the TX side since the same problem happen with multiple RXs.  I have installed and re-installed the same versions, and can't duplicate the disconnection scenario.

glipski50, 2024-10-07:
Disabling DEBUG did not help.
Attached the log file from just after the error mediate and a brief video. Just after clicking "Forward Programming RX" the error message shows up very short. Best is to scroll forward manually.

[Uploading Archiv.zip…]()

frankiearzu, 2024-10-08:
The iniital "No compatible RX" is ok to show quickly in the beginning... is for a little bit while the RX reponds to our request to get the RX version.

Navigate the menus at a normal pace.. once the menu/values show up, the select the options.  returning too fast created disconnects in the past. 

Can you do a run where it disconnects, with the DEBUG_ON=1 to write to the logs..

glipski50, 2024-10-22:
Sorry for beeing a bit late. I only experienced „No compatible RX“ at the beginning. I did some more setup using the SM Tools but I did not experience a disconnect during the activities in the menues. 

I keep DEBUG_ON=1. Should I get a disconnect DURING menue activities I will send the log.

HectorHL, 2024-11-22:
Having same issue. 
TX16S (First Editions) on EdgeTX v2.10.5
DSM Forward Prog v0.59 (Clean install)
AR637T v3.1.3

frankiearzu, 2024-11-22:
@HectorHL thats exactly the setup I have on my desk.. same versions.
Make sure the bind is DSM X2F, and that you are receiving telemetry.

closed 2024-12-16 by frankiearzu