frankiearzu/DSMTools #29: Relearn Servos does not adjust gyro direction

chadgms, 2024-10-13:
I setup my Eritax with a from scratch loading a blank new profile from the Spektrum programmer.   I was able to setup everything, but when testing the gyro directions I found that it was applying down elevator when I pointed it down, so it did not know the servo was reversed.   So, I did the relearn function and it did not help.    

I re-bound it to my NX8 and did the relearn and that fixed it.   re-bound it to the TX16 and it was working fine.   So, looks like it does not get the proper information from the servos.

qsiguy, 2024-10-13:
It works, done if dozens of times.  You have to go into the model setup in the "pre-forward programming" menu in the script.  You reverse the channel direction there, then go into forward programming and do the relearn servo settings.  If you only reverse the direction in the radio model setting in outputs, it doesn't send that to the receiver during the relearn procedure so just because your radio moves it correctly, the gyro isn't updated.  The model files in the /MODELS/DSMDATA folder get edited when you edit the model from the script and that is the data sent to the receiver.

chadgms, 2024-10-16:
Ah, got it.   I thought it would pull from the model outputs.   I see where you specify this now.   Thanks!

closed 2024-10-17 by frankiearzu