frankiearzu/DSMTools #30: Forward Programming and Telemetry not working on Taranis

10man, 2024-10-25:
Trying to use the DSM tools to setup Blade helicopters.

Radio: Taranis X9D+2019
EdgeTX: 2.10.5
DSM Tools: 0.59 (MIN)

Forward Programming:
    * Setup script has no options for helicopter on version 0.59 using Taranis X9D+2019. Without a setup, the min script fails to find a model config.

Telemetry:
    * On SmartRX (Blade 330s running 5.5 firmware) not getting any menus populated
    * On 636 Receiver (Blade 230s v1 running 4.5) no text-gen is shown.

frankiearzu, 2024-10-26:
Configure the helicopter as a regular plane.. Ail/Ele/Thr/Rud.. the Heli receivers don't use any of the plane info. We just need the to create a config.

Also the X9D could have too few memory, and you need to use one of the older versions.  If it fails the first time, run it again (this is to compile the lua code), then restart the TX, and run the script.. that is when it does have the more memory available.

What RX does the Blade 330 has??  FH6250H??   If it is the one based on AR636. will probably not have forward prog.

On the 230s V1 with the AR366, the latest firmware that is intended for the RX with aluminum case does have limited forward prog. Others does not.
TextGen should work in this v1... are you using the DSM_SmartRX_Tel script, and menu option TextGen??  
  
I used to have a 230s V1, but not any more.. but remember that I used a version with limited FP and had TextGen for servo subtrims and other small functionality.. at least should show the starting text with the receiver name, then you have to use a combination of stick movements + panic to enter the specific TextGen config.

10man, 2024-10-28:
For the 230s (636 receiver):
- the receiver is actually SPMAR6115E
- telemetry works as does adjusting servo centers & a3x gains
- no text gen, perhaps the esc needs an update..?
---
For the 330s:
- the receiver is SPM4649T
- telemetry connects (rf signal and bec volts) but nothing updates in the telemetry data script
- no text gen, perhaps esc needs an update..?

frankiearzu, 2024-10-28:
Just making sure that you are using DSM  X2F subprotocol... the other ones usually don't have telemetry.

On the 230s V1, what i remember is that the textgen is not for the ESC.. is to adjust subtrims in the servos, and gains.   The Avian ESCs with TextGen only works with the new Smart Line of receivers.. AR630/631/637, etc..  who supports SLRX2. (not AR636).   
Found one of my earlier videos, at 6:40 you can see what is provided for the V1 firmware version 4.40
https://www.youtube.com/watch?v=yrAqV4b9Li4

On your  330s, SPM4649T is an older satelite receiver... any textGen or Forward programming will be provided by the Flight controller.. 
On my Spektrum helis, I have the flight controller is FC6250HX, and a SLRX2 serial receiver SPM4651T.       
Forward Prog is provided by the FC6250. no TextGen.
What flight controller are you using??

10man, 2024-10-28:
330 flight controller is SPMFC6250HX and I'm pretty sure it's running an avian esc. I will verify.

The info on the 230 in terms of version and receiver model may be wrong. I think I had something funny in my spectrum account... I will check again.

frankiearzu, 2024-10-28:
Ok, the FC6250HX has forward prog... wondering about the Satelite RX that you have.  i have the SLRX2 model SPM4651T.

frankiearzu, 2024-10-28:
On the 330s, you should be getting some telemetry from the RX.. If you do "discover sensors", it should get also info about the Avian ESC: eRpm, eCur, vIn (flight pack voltage), etc.        Im thinking that with an older RX, the FC is not updating... i think i have an RX that is not SLRX2.. will try that later.     As for the format, spectrum uses a standard publided format for their telemetry... there is only one.. they populate both the Telemetry Sensors, and the same data is used in the LUA scripts.. so if you are not getting more telemetry sensors, the LUA will not show them,

closed 2024-12-16 by frankiearzu