frankiearzu/DSMTools #32: Info error

electric-2000, 2024-12-15:
I get this error under the Info page.

[image]

electric-2000, 2024-12-15:
I think I figured out what created this error. I keep my models in sub folders under the Model folder. When setting up a plane in DSM Tools I could not save a configured plane. After sleeping on it overnight, it dawned on me that I'm using folders. I moved my model.bin to the roof of the Model folder. Now I can configure a plane and it saves to the proper DSM Tools folder.

Bottom line... DSM Tools does not like using sub-folders under the Model folder.

frankiearzu, 2024-12-16:
Didn’t know that sub folders are supported in EdgeTx inside /models.That’s why the have categories/tags/models/DSMDATA is where the tools save the forward prog data for a model using the model name.

frankiearzu, 2024-12-16:
Ok,  was able to duplicate your error. yes, is a bug on my side.

The "data" path where model-setup stores the settings is relative to the script path.. "/scripts/dsm-tools/data".
So if you have a plane in a sub-folder of model, like /model/glider/test.bin I'm trying to write the config to: /scripts/dsm-tools/data/glider/test.txt    that folder "glider" does not exist under my script paths... I have to create it.

Will fix it, and will post it here when ready... give me a few nights!!
Thank you for the testing!!

frankiearzu, 2024-12-16:
Fixed.. download the RC2 zip file..   just copy it on top, it will replace the older files.  
What i am doing is that i am creating the same sub-folder inside my data folder (/scripts/dsm-tools/data/<subFolder>)

Also related to the screen resolution, it will default to the lowest resolution if is not in my UI config instead of reporting the error.
Let me know what radio you are using and will adjust the UI accordingly for that Resolution

electric-2000, 2024-12-16:
Thanks for the quick work. I'm using the X14S and I will give it a test tomorrow.

-Mike

electric-2000, 2024-12-17:
I used Ethos Suite to upload the Lua. Your RC1 was previously installed, and it overwrote the files with no problem. The model located in a subfolder worked great...no more errors. Looking forward to seeing today's updated on Ethos.

closed 2024-12-29 by frankiearzu