frankiearzu/DSMTools #36: "No compatible DSM RX.."

gardelf, 2025-01-23:
Hello everybody,,
I have a tx16s  with edge Tx, last firmware 2.10.2 
When I access to Forward programming , screen give this message "No compatible DSM RX..". 
Receiver is a new ar630. 
Do i have to do something with this receiver to access to Forward programming...?¿
Thank you in advance for any response...

gardelf, 2025-01-23:
Hello again, thank you for your quick response
I ve  updated FP and Firmware as you told me.
At the moment I try to access to FP,  receiver is bound to the tx16s, but doesn't work..
I have  discovered the issue, it was version internal Module, I have updated to [mm-stm-serial-aetr-air-v1.3.4.0.bin] AETR in my case
Thank you for been there
Fernando

frankiearzu, 2025-01-23:
Fernando, 

The AR630 is supported. Do you know what firmware version??

The message "No compatible DSM RX" really means that the TX sent a command to get the FP version of the RX, and the RX did not respond.   The TX sends the commands as part of the stick positions information, and the response from the RX comes via the telemetry stream.

Possible causes:
1.  No telemetry (communication back from RX to TX).  Validate that if you do "discover new sensors", you can see the values changing.   If not, check that the telemetry is enabled in the Internal/External module. Also, check that the sub-protocol is DSM X-2F to bind (Auto should negotiate to that)
2.  Some BNF receivers are locked and need the latest firmware version to unlock.  With the USB dongle, the firmware can be updated.

closed 2025-02-01 by frankiearzu