frankiearzu/DSMTools #37: X14S Screen issue

electric-2000, 2025-02-03:
At one point you mentioned fixing the screen issues for the X14S. Currently using dsm-tools-ethos-2.0-final.

When running the Aircraft setup:

The Tail Type selection is hidden under the Next button. 
THX

-Mike

frankiearzu, 2025-02-10:
For the X14, seems that I'm using fonts that are too big for this screen.

If I change the fonts from Standard to Small, seems that it fits better.
If you want to make the changes to your installation,  This is the part that controls the font sizes for the different radios.

The file to edit is:    /scripts/dsm-tools/lib/lib-ui.lua
[image]

After those changes, it looks better in the X14S sim.  Will create a 2.0.1 release in a few days with other cosmetic changes.

[image]

frankiearzu, 2025-02-10:
If you want the entire zip, is here:
https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Tools%20Suite/dsm-tools-ethos-2.0.1-RC1.zip

electric-2000, 2025-02-10:
Thank you for the quick update. I'll D/L'd it and put the Lua script in my radio today.

https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Tools%20Suite/dsm-tools-ethos-2.0.1-RC1.zip

closed 2025-02-18 by frankiearzu