frankiearzu/DSMTools #38: Cannot access forward programming

modelist1, 2025-02-13:
Using ar637t and v2.0 on a x20pro ethos 1.61. When entering forward prigramming i get: loading message file: 0

frankiearzu, 2025-02-15:
That's the part where it is loading the message file..
The message file is in the folder/directory:   /scripts/dsm-tools/i18n/msg_fwdp_en.txt

Can you verify that the file exists?

is your /scripts folder on the flash or SDCard?   this script needs to be on the SDCard.

modelist1, 2025-02-15:
That file is that location. Before raising this issue i already checked and even reinstalled that txt file

modelist1, 2025-02-15:
It is indeed at the location. Even tried copying it again

[image]
[image]

frankiearzu, 2025-02-15:
I have my /scripts (all lowercase) you have /Scripts (1st uppercase).

Don't remember if ethos is case sensitive. Not at home but will try later.

modelist1, 2025-02-15:
Same happens if I change it to lower case. The part of the tools marked telemetry works fine in both cases and I can read as3x or esc data for example.

modelist1, 2025-02-17:
When trying to use fp the warning triangle (see picture) appears

frankiearzu, 2025-02-17:
The red triangle means an error.. 
can you go to sys->info where firmware version is shown, and also will show a description of the error for the red triangle.
Can you post a pic.

modelist1, 2025-02-17:
[image]

frankiearzu, 2025-02-17:
Thats strange.  The seek function is very standard. I just trying to go to the beginning of the file.
 
Working on re-writing the loading of the messages to get them all much faster, and will not use that "seek" function.
Will finish it tonight.. will post it here when is ready.

frankiearzu, 2025-02-18:
Can you try v2.0.1-RC2 please... complete rewrite the way I was loading messages to make it faster than before, and not using the "seek" that was giving you error.

Ran on my X20RS and in PC simulators

https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Tools%20Suite/ethos/dsm-tools-ethos-2.0.1-RC2.zip

modelist1, 2025-02-18:
Great job. I am currently working with it and will update this message if I find more. 
1 put it on sd but dsm icon did not appear (read script was pointing to sd)
2 put it on radio and changed pointer to radio: got dsm icon plane configured and now can enter fp. 
3 small issue, no back button in main menu.

frankiearzu, 2025-02-18:
Can you explain your #2? You changed the "dsm-tools/main.lua"  to point to "radio:...." on the folder/directory paths? or somewhere else?

The FP back button.. The menus come from the RX, including the info to display "back" or not, but the "RTN" key always works.  I have to hack it to add it just to the main FP screen  (Override what the RX is sending me, and handling different to return to my script).

The Flight mode, that sometimes it shows invalid values temporarily..   What happens is that the validation for values is done at the RX.. the screen only advances to the next or previous option assuming that is correct, but if the RX knows that the option is not valid, it changes it to the valid one.. thats why it gets corrected after 1/2 second later.  Imagine that the list has "Option1, Option2, Option3, Option4".. only "Option1" and "Option 4" are valid for the current RX settings.. If the menu is displaying "Option 4" and we do previous, I assime that is "Option 3", but the RX later tell me that "Option 1" is the only valid previous option.    On genuine Spektrum TX the response times are really fast.. you can still notice this behaviour if you are putting attention..  Here with LUA and the multi-module, its a little slower and you notice this a bit more.   This behaviour happens everywhere where you have non-numeric options.

Well, good that it finally works!

closed 2025-02-22 by frankiearzu

BladeScraper-Designs, 2025-02-27:
Commenting to point out that [this](https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Tools%20Suite/ethos/dsm-tools-ethos-2.0.1-RC2.zip) fixed issues for me as well.

With 2.0 release, I had this when trying to go to fwd prog:
[image]

frankiearzu, 2025-02-28:
Looks like the "seek" problem is only on the X20RS.  The 2.0 version works well on the on others and the X20 simulator.