frankiearzu/DSMTools #39: Radiomaster Boxer FM Mode not changing - Only shows FM Mode %s

WhatevsJJ, 2025-02-19:
I have CH8 programmed as FM channel but when I flip the switch assigned to CH8 Flight Mode never changes beyond the %s.  I do see CH8 moving in the output window on TX so believe the radio is set up properly.  Perhaps TX CH8 isn't RX CH8?.?.?

Thank you

[https://photos.app.goo.gl/JjFVMeXGXiQJM5HE6](url)

qsiguy, 2025-02-19:
Make sure in the model RF setup you enable channels 1-12. They will often default to 1-6 and then nothing higher than channel 6 will be transmitted to the receiver.

WhatevsJJ, 2025-02-19:
Good question, but yes, I have 1-16 configured.  Here are a few more pics.

[Pics of model & channel config](https://photos.app.goo.gl/oe4hvmnF6J2h6evA7)

frankiearzu, 2025-02-19:
Just tested on the boxer.. works fine with me. Using version 0.59.If is showing %s, I think you have a mix of versions. Probably reinstall from zip.. just make sure you delete the SCRIPTS/TOOLS/DSMLIB and the previous version of forward prog in SCRIPTS/TOOLS

frankiearzu, 2025-02-20:
[image]

frankiearzu, 2025-02-20:
Was able to reproduce the %s problem.. let me fix it tonight

frankiearzu, 2025-02-20:
Found the problem.. forgot to update one file on the B&W zip file.
Please use this one: v0.59b

If you have already version 0.59 or 0.59a, you just need to copy on top of what you have, it will override the existing file.

https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Forward%20Programming/Black_White_128x64/SD_FP_59b_MIN_Feb19.zip

WhatevsJJ, 2025-02-20:
> Was able to reproduce the %s problem.. let me fix it tonight

You are amazing - I was just about to post I am seeing the same even issue after replacing all files.

Thank you!

WhatevsJJ, 2025-02-20:
> Found the problem.. forgot to update one file on the B&W zip file. Please use this one: v0.59b
>
> If you have already version 0.59 or 0.59a, you just need to copy on top of what you have, it will [...]

That fixed it, thank you again.  Now I'm trying to figure out why CH8 wont actually change modes... : (

frankiearzu, 2025-02-20:
Just do ch1-12 range.. not 14 or 16.Also make sure that you have in the module (where you bind), extended throws OFF (or similar).

  

https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Forward%20Programming/Black_White_128x64/SD_FP_59b_MIN_Feb19.zip

https://github.com/frankiearzu/DSMTools/blob/main/DSM%20Forward%20Programming/Black_White_128x64/SD_FP_59b_MIN_Feb19.zip

qsiguy, 2025-02-20:
> Good question, but yes, I have 1-16 configured. Here are a few more pics.
> 
> [Pics of model & channel config](https://photos.app.goo.gl/oe4hvmnF6J2h6evA7)

You don't want 1-16, needs to be 1-12.

WhatevsJJ, 2025-02-20:
> Just do ch1-12 range.. not 14 or 16.Also make sure that you have in the module (where you bind), extended throws OFF (or similar).

frankiearzu - How can I compensate you for helping me and putting this valuable resource together?  Paypal address?

frankiearzu, 2025-02-22:
Yes, Avian ESC is programmed using TextGen.  It is part of the Telemetry tools. (smart RX telemetry)

https://www.youtube.com/watch?v=4AHG5nsJX_Y&t=165s

closed 2025-03-02 by frankiearzu