frankiearzu/DSMTools #40: [Ethos] Capture Data pages not displaying correctly on X20RS

BladeScraper-Designs, 2025-02-27:
Not sure if this issue is exclusive to the X20RS or if other radios do it too.  I just finally got dsm-tools to work (turned out to be an issue with my module) and just started using it and noticed this.

[image]
[image]
[image]

frankiearzu, 2025-02-28:
Many radios with different resolutions need some adjustments.  Will fix it for RC3.    Capture is not critical. The idea of it is if someone has a telemetry sensor that I don't have,  we can capture the data (stored in the dsm-tools/logs folder), and I can use it to reverse engineer it and add it to the simulator for testing.

frankiearzu, 2025-03-02:
Fixed on 2.0.1 final

closed 2025-03-02 by frankiearzu