frankiearzu/DSMTools #41: Radiomaster GX12 - Forward Programming fails to execute

makaluch, 2025-03-24:
Forward programming fails to execute on Radiomaster GX12. *Unknown Error*

Forward Programming: 59b_min
EdgeTX ver 2.11.0-RM-build
Ext TBS 4n1 Module ver: 1.3.431
Spektrum AR631 ver: 3.2
Bound: X 2F 11ms

frankiearzu, 2025-03-24:
I will need more info.  It it failing to start?
 
Can you check that the /MODELS/DSMDATA directory is there?
Do you have the /SCRIPTS/TOOLS/DSMLIB directory there?

The /SCRIPTS/TOOLS should have this:
[image]

and /SCRIPTS/TOOLS/DSMLIB
[image]

Did you run the model setup LUA before running forward programming?
There should be in the /LOGS/dsm_min_log.txt, can you attach it here.

frankiearzu, 2025-03-26:
I think i found a problem with 2.11.0-RC3..    Is not liking long filenames.   If you rename the files and use "FP" instead of "FwdPrg" it stated to work on my Boxter radio.

My /SCRIPTS/TOOLS/    directory now looks like this:

[image]

frankiearzu, 2025-03-27:
Just fyi, a fix for the long filenames has been done..  Will be on the next release (2.11.0-RC4).
In the meantime, shorten the filenames.
https://github.com/EdgeTX/edgetx/issues/6026

closed 2025-04-09 by frankiearzu