frankiearzu/DSMTools #42: "No compatible DSM Rx" error in forward programming.

degooglified, 2025-04-23:
I have the following config:

Jumper T18 Pro tx
DSM Fwd Prog v2.0
edgetx 2.10.6
Internal and internal Multi module running V1.3.4.31 TAER

The receiver is a AR637TA, unlocked running latest firmware , updated in March.

The receiver binds well and telemetry works fine as does the setting of the AS3X modes.

Suggestions to solve this?

Tx in adv.

Geert.

frankiearzu, 2025-04-23:
You are using the "suite" version 2.0..   not to be confused with the really old v0.2

That error also means that the RX is not responding to a request to get the menu for forward programming.
The menu is stored in the RX, and we are trying to get the main menu.

1. Validate that you bind with DSM-X2F
2. On the telemetry page, do discover sensors..   That is to validate that the RX can send data back to the RX.

If that does not solve the problem:

If is an older BNF, sometimes they come "locked" so that you cannot modify the settings.
to solve that, you need to upgrade the RX to the latest version with a PC with a USB cable.

degooglified, 2025-04-24:
X2f is used.
Throttle is low
Discovery of telemetry sensors works well

The receiver has been unlocked.

Fwd programming used to work very well with this receiver and it was unlocked already at that time.

I

frankiearzu, 2025-04-25:
Do you happen to remember what version was working for you before??? 0.59?  The suite 2.0 is mostly a re-package of the 0,59a.   im wondering if it worked before with you, if that is something that was incorporated in this version.

Do you have any other RX??  This is to test if it is the script or just something with this RX.

Also, can you send me the log file of your last run?? is stored in /LOGS/dsm_log.txt, i just want to check if there is any error there.

degooglified, 2025-04-25:
It used to work well when I was in edgetx 2.10.5 and mt-t18int-TAER-lbt-v3.4.0  and the 2.0 DSM tools suite (not 0.59). (i sent you a donation to thank you for the good work ;-) )

And then it stopped working with the above config, without changing anything, except the addition of the external module in the T18.

So I decided to upgrade to the latest software:

edgetx 2.10.6
Internal and external Multi module running V1.3.4.31 TAER ('air' image)

I have another DSMx RX, but not one with forward programming capability.

Log enclosed.

[dsm_log.txt](https://github.com/user-attachments/files/19904968/dsm_log.txt)

frankiearzu, 2025-04-26:
Just saw that had a bug with the logs... was truncating/deleting it at exit.   Can you use the new ver 2.0.0a for color.

https://github.com/frankiearzu/DSMTools/tree/main/DSM%20Tools%20Suite/edgetx

The log file is readable with notepad or other text editor.

When trying to connect,  it is trying to get the version of the RX... if it can't, you will see the "No Compatible DSM RX" until it gets the firmware version.   Forward programming works by transmitting a request to the RX using the signal Channel, and the responses via telemetry.. thats why we try to validate that you are receiving telemetry.

The only difference with my setup is that i am using MM v1.3.4.0 AETR (air), but will try the one you are using..  

The normal operation will have something like this:
[image]

When it cannot get the version from the RX, will look like this.. keeps trying to get the version

[image]

degooglified, 2025-04-26:
log enclosed with 2.0a tools

[dsm_log.txt](https://github.com/user-attachments/files/19921207/dsm_log.txt)

frankiearzu, 2025-04-26:
Is not getting anything from the RX.. running out of ideas.. Have you tried with another RX? or re-flash the firmware in that RX?

qsiguy, 2025-04-26:
In your Model RF setup do you have channels 1-12 enabled?

frankiearzu, 2025-04-26:
Looking at the log, he has the MM configured as TAER,  using 12 channels.  I tried last night the TAER MM firmware, and it did worked the same way as my regular AETR.  He is using DSM-X2F that is correct.

dsarnson, 2025-06-28:
Guys also having this problem. Got a bnf plane, which had already been bound to a spectrum TX, and trying to get setup with my TX16S
AR637TA
Edgetx 2.10.6
MultiModule 1.3.4.31 AETR
Binds well and confirmed DSM X 2F  Ch 1-12.
All control surfaces work well with sticks
Downloaded and copied files into appropriate folders on SD card but when I try to run  DSMFwdPrg59_color, I also get the dreaded "No compatible DSM RX.."
Do I need to reset the RX or 'unlock' it somehow as it came in an Eflite plane, probably already setup for Spectrum.  I would think rebinded to my TX would do that. 
Any other thoughts on trouble shooting??
Cheers.

dsarnson, 2025-06-28:
Thanks, so I deleted all the DSM stuff from the SD card, and added the DSM Tools Suite..... bound the rx again, and all surfaces work, but still get the 'No compatible DSM RX' when trying to run the lua.    ???

qsiguy, 2025-06-28:
This isn't an issue with v0.59 vs the new Suite as the Suite contains v0.59 just repackaged into the Suite with the other tools.

dsarnson, did you previously connect via Forward Programming with your Spektrum radio?  If so, do you recall which firmware version you have on the AR637TA?  Some older firmware versions with certain BNF oem programming will not accept Forward Programming connections and it would not matter if you were using a Spektrum radio or EdgeTX on your TX16S.

On the TX16S did you first run the model setup to ensure the DSMDATA file was generated for that particular model then attempt connection to Forward Programming?

ror1948, 2025-06-28:
Also does enable max throw have to be off.Bind with auto and it show pick dsmx

dsarnson, 2025-06-28:
Thanks for that.....unfortunately I do not know the former setup as this was purchased as a 2nd hand BNF.  It WAS working with the former Spectrum Nx8 for demo purposes, but I don't know if he had configured safe/as3x or not, it was rarely flown.  I don't know the AR637TA fw version either.  Maybe should figure out how to update RX fw.  And yes, I was able to run the model setup and a DSMDATA txt file was generated, and then I tried FP without luck.  
And Enable max throw is OFF.  I did several binds, first with Auto, and it defaulted to X2F.  
Maybe it IS the RX fw version.....???

ror1948, 2025-06-28:
yes if it is a stock BNF you can not fully FP it.had the same the other day on a eflite commander .but the FP did allow you to pick the SAFE channel.At first it said no Rx until i installed the latest DSMtools

qsiguy, 2025-06-28:
"Enable Max Throws" must be off for the f-modes channel to properly switch between safe/as3x modes but should not prevent access to Forward Programming.

dsarnson, what airplane is this on?  Some trainers will have preset channels for safe modes and panic that you cannot change and may not have access to Forward Programming without unlocking the receiver and doing a factory reset, the AeroScout is an example of that.  Newer AeroScouts have some access but not full access.  Other BNF airplanes configured from E-Flite as "SAFE Select" you will have limited access to Forward Programming but do have the ability to select the f-modes (safe/as3x) channel.  When you power on the plane and let it sit undisturbed for a few seconds does it "wiggle" the control surfaces 2 times upon initialization?  If so, that means safe is enabled and you just need to map a switch to the correct channel to toggle it.

If it is a SAFE Select plane, you can map your desired SAFE mode 3-way switch to any channel, 5-9, and if your sticks are moving the control surfaces in the correct directions, you just have to hold the sticks down and in and flip the configured safe mode switch (I generally use the SA switch) 5 times.  On a safe select plane this will assign that channel as the f-modes channel.   Make sure the "Enable max throws" checkbox is NOT turned on in your model RF setup.

qsiguy, 2025-06-28:
Hmm. I have no personal experience with the Commander although I do with many other planes.  The Commander manual states it is programmable with the "stick down and in/switch flip".  Hard to say for sure what the issue was.  Any that have the ability to program with the stick/switch method (which is SAFE Select) also have the limited access through forward programming to assign the channel as well.

dsarnson, one more thing to check.  Bind up to the model and go to the telemetry screen and perform the "Discover new" procedure to ensure you have the telemetry link established. It uses telemetry to make the connection so if that isn't setup it will not work.

If still no luck, see if you have a dsm_log.txt file in your radios SD card in the /LOGS folder.  Copy/paste the log here so we can analyze it.  If you have the log it will usually contain details of the DSM communication and tell us why it failed to connect.

qsiguy, 2025-06-28:
Channel 7 mixes assign the SB switch as the source. Do NOT put anything in the "switch" field.

Bind up to the plane, do NOT advance the throttle yet, and with the SB switch in position 0 (full up or full down is OK) pick up the plane and move it side to side and see if the ailerons attempt to auto correct and level the plane.  If they do not, flip the SB switch to the opposite position, not center.  Repeat the test moving the plane.  If safe is on that channel in one of those positions SAFE auto leveling will be active.   AS3X only activates after you advance the throttle over 25% but only moves the control surfaces a small amount with the plane is wiggled as if from wind.

If channel 7 did not do anything, I assume channels 5 and 6 are gear and flaps?  Does the plane have retracts?  Move the SB switch to channel 8 and repeat the tests.

If you want to assign channel 7 as your safe mode switch, while SB is assigned to channel 7, plane on and bound, hold your sticks down and to the inside and flip the SB switch 5 times.  (OFF->ON->OFF = one "flip").  That should remap channel 7 to be the f-modes or SAFE switch.

dsarnson, 2025-06-28:
I did 'Discover New' Telemetry, and tons of fields populated, so that is working.  Tried FP again to the same error.  Here is the content of the txt file in the DSMDATA folder titled P51_S11065.txt.   Interestingly I titled the model "P51 Spectrum"  
1001:0
1002:0
1003:1
1010:0
1011:1
1015:3
1017:2
1020:0
1021:1
1022:0
1023:1
1024:1
1025:0
1026:0
1027:0
1028:0
1029:0

qsiguy, 2025-06-28:
The DSMDATA file is only used when configuring the model in Forward Programming so it selects the correct channels, channel direction, wing type, etc.   Here is what those values mean.

1001:0 | Aircraft type (0=Airplane)
1002:1 | Wing type (0=single aileron, 1=dual aileron)
1003:1 | Tail Type (1=normal 1 rud+1 elev)
1010:0 | Throttle/Port1 ch assignment (0=ch3)
1011:1 | Aileron left/Port2 ch assignment (1=ch1)
1012:4 | Aileron right/Port2 ch assignment (4=ch5?)
1015:3 | Rudder/Port4 ch assignment (3=ch4)
1017:2 | Elevator/Port3 ch assignment (2=ch2)
1020:0 | Throttle ch 3 (1=reversed output)
1021:1 | Aileron ch 1 (1=reversed output)
1022:0 | Elevator ch 2 (1=reversed output)
1023:1 | Rudder ch 4 (1=reversed output)
1024:0 | Ch 5 (1=reversed output)
1025:0 | Ch 6 (1=reversed output)
1026:0 | Ch 7 (1=reversed output)
1027:0 | Ch 8 (1=reversed output)
1028:0 | Ch 9 (1=reversed output)
1029:0 | Ch 10 (1=reversed output)

qsiguy, 2025-06-28:
No, his had single aileron.  I didn't use his values, just an example I already had I could copy/paste easily.  More about the "legend" on the right to decipher your file.

dsarnson, FYI that DSMDATA file name syntax is normal, don't rename it.  Not assuming you were going to but just want to make sure.

dsarnson, 2025-06-28:
Things are weird.  I am using high rates, and have ch7 assigned to Sw SB.  Ch 5 is gear and Ch 6 is flaps.  When powered on, I can tilt the plane and the ailerons and elevators move to 'auto-correct', but that situation is active no matter what position the SB switch is in!  I also tried with the sticks down and in, then flipping the SB switch 5 times.  I am trying to have one position off, then AS3X, then Safe/SelfLevel.  But it seems to be acting like it is SelfLevel mode all the time.  I did read about a 'panic' setting that might be on another channel than might have to be assigned?  So confusifying! (to quote a Wicked character!)

qsiguy, 2025-06-28:
In your telemetry list of values do you have one labeled "FM"?  That is your flight mode and rather than relying on checking if the gyro is on, you can just watch the FM value and see if it cycles through 1, 2, & 3 as you move your switch.  If you get that responding and if auto leveling appears to still be active while in mode 3 "AS3X" it may be in panic mode.  In that case I would leave it in FM 3 and then try forcing the other channels to +100 to see if auto leveling (presumably panic) turns off.

Here's an image of my telemetry list showing "FM" while in mode 3 AS3X.

[image]

frankiearzu, 2025-06-29:
@dsarnson   how many channels do you have configured on the page where do bind??  Try Ch1-Ch6, and Ch1-Ch12 and re-bind.

I haven't tried the latest version intended for AS3X+ on older versions..  Will try that.. There was a field in the connection settings that changed for AS3X+.   Will downgrade one of my test receivers to check that it works with older versions of the firmware.

frankiearzu, 2025-06-29:
I have an AR637 with an older firmware version 2.38.1 (from around 2021) and downloaded the EFlight P51 AS3X settings.   The DSMTools suite worked with the older version.

My settings on the Internal RF tab:
Multi-Module v1.3.4.31
DSM X2F
Enable MAx Throw OFF
Servo Update Rate 22ms
Ch Range  Ch1-Ch8.   

The Eflight P51 settings from Spektrum manual:  (1.2m wingspan)

By default, it does not come with a switch to change flight modes. Only Safe Select ON/OFF via the bind plug.
They usually do this when is a 6 ch plane (Retract + Flap).

Configure forward programming plane setup single Aileron, and forget about the flaps at this time.. they are not used for AS3X/Gyro anyway. 

Wing Type: Single Aileron   (connected via Y)..   NOT Dual Aileron
Tail Type: Normal.

Ch1=Thr
Ch2=Ail
Ch3=Ele
Ch4=Rud

On EdgeTX, you have to set up everything as: (Assuming that the multimodule will do the Ch-Map from AETR->TAER).
Ch1=Ail
Ch2=El
Ch3=Thr
Ch4=Rud

Ch5=Gear/Retract
Ch6=Flaps
Ch7=Flight Mode Switch (source is a 3-pos switch)

Be default, is locked, but still allows me to do SAFE SELECT via forward programming.
I was able to Setup the flight mode via forward programming on Ch7.

qsiguy, 2025-06-29:
Ultimately if you update the firmware, unlock, and reprogram it from scratch you'll be much better off once you get it dialed in and have full access to customize it.

As for the "stick/switch" method as I mentioned previously one "flip" = OFF->ON->OFF.  Middle position is not relevant so go from fully off to fully on and back to fully off for one flip. Hope that makes sense.

txkflier, 2025-06-29:
What is the model number of the receiver?

If it's an AR636A, the receiver's flight modes channel can be set by assigning a switch to a channel (other than channel 6), holding the sticks down and toward each each in 100% rates, and cycling the switch 5 times. A cycle is one way and then the other. Panic mode is hard coded to channel 6, so you'll need to assign a switch to channel 6, preferably a pushbutton.

If it's a Smart AS3X/SAFE receiver, it won't have Panic Mode and you can assign a channel to its flight modes using the procedure above.

dsarnson, 2025-06-29:
It is the AR637TA. 
As for the 5 flip switch cycle.   I did it yet again, but _really_ fast, and now that 3 position switch does actually work....sorta.....I can now choose between AS3X and Safe/Self-level modes, but still not able to completely turn off.  The telemetry screen does show the switch changing FM between AS3X and AS3X-Level.  
I may try to fly and see how it responds, but not really comfortable without the ability to disable gyro.  Or I may just unlock, reset, and initiate setup.......
Stay tuned

qsiguy, 2025-06-29:
Default BNF receiver setup won't have a gyro OFF mode.  Even if you bind with safe off, all modes will be AS3X.  If you decide to factory reset it and reprogram, you can then pick one mode to disable the gyro completely.

Great news on getting the channel assignment done!

Shane
-Shane's DIY Channel

frankiearzu, 2025-07-03:
As Shane said, many BNF don't have the option to completly disable it.   The FM sensor helps a lot to troubleshoot flight modes.

Good discovery that even with a Spektrum TX, you were not able to do forward programming.. They only allowed locked RX to be "unlocked" via forward programming until about 1-2 years ago.    Once you update the firmware with the USB cable, you should be able to program it completly.

txkflier, 2025-07-03:
According to the AR637T change log, Version 2.37.8 (October 6, 2020) was the first version that allowed you to factory reset a BNF receiver. All I have available at the moment is an AR8360T and the oldest version it will let me install is 2.38.5 (December 21, 2020). In its SAFE Select menu, it will let me turn AS3X off in either flight mode 2 or 3 or both of them. If I enable the SAFE Select option, flight mode 1 is SAFE mode and the option to turn AS3X off is hidden since SAFE needs AS3X to function properly.

dsarnson, 2025-07-03:
I see this is now closed, but I did want to add a quick update, that upon updating the FW to 2.48, the rx was unlocked and allowed Forward Programming and factory reset.  Started from scratch with now configured AS3X, Off, Safe on a 3 position switch.  Maiden this weekend.  Thanks all for the comments!

txkflier, 2025-07-03:
Good deal. Be aware that the angle limits in SAFE mode won't be correct unless your dual rates are set to 100%.