frankiearzu/DSMTools #46: Confirm Spektrum RSSI/TRSS telemetry to EdgeTX is not reliable?

dsarnson, 2025-07-12:
Guys, you may recall, and I thank you for all the help getting my Spektrum BNF AR637TA flashed to newer firmware 2.48 so that Forward Programming now works.  Bound DSM X2F and everything setup nicely.

I did a low power range check from very far away, and all seems good.  I discovered all the telemetry, and when I fly, I consistently get "RSSI Critical" voice alerts from my TX16s running EdgeTx 2.10.6 with MM V1.3.4.31.

The TRSS as reported in logs starts out about 31 and rapidly drops out sometimes to below 10.

Searching around elsewhere on the web, it looks like Spektrum telemetry regarding RSSI/TRSS is not reliable, and the only remedy is to either "Disable telemetry alarms"  or set the Low alarm and Critical alarm to crazy low numbers (right now they default to 45 and 42).

The 'fade' telemetry value of FdeA goes from 0 to less than 50 over a 4 minute flight.  I think that is ok, right?

Cheers, DSA

ror1948, 2025-07-12:
Yes you have to disable telemetry alarms on DSM.RSSI is meaningless on Spektrum...All the more reason to switch to ELRS or Crossfire

dsarnson, 2025-07-12:
Thank you for the confirmation.  I usually fly FrSky stuff, been rock solid.  The only reason for Spektrum, it came installed in the plane that I got from my clubmate.  I was up for a challenge to get it to work.  It works well now, but I might swap it out for FrSky and sell it......
Cheers.

frankiearzu, 2025-07-12:
The TRSSI is a value in the range of  0-31, and to get the 100% graph RSSI, the TRSSI is multiplied by 3, so it changes fast.

The AR637 is full range telemetry, so the alarms, even not very accurate, are still usefull.  i drop the default alarm values at least 10 points (Low=35, Critical=30) or even smaller if i get too many warnings.. i get low warnings once in a while on the cornes of the field.  

The AR630/631 and UMX are short-range telemetry, called "fly-by".. for those you will loose telemetry at about 50-70', you have to turn OFF the alarms. Otherwise it will be anoying.

For the other sensors **FLss** (Frame looses) is the important one... acceptable is 100 per minute of flight (still > 98% quality).
**Holds** are how many times it has gone into failSafe.. that one you want it to be always 0.

closed 2025-09-03 by frankiearzu