frankiearzu/DSMTools #52: GYRO Reverse

blue68camaro-git, 2025-09-02:
Not sure why I can not reverse a GYRO setting. Utilizing an AR636, FrSky X20s, Latest DSM Tools, Latest TX software. I set the Ailerons to reverse but they never reverse when checking GYRO movement. 

The Statement Says inGRO Reverse(PORT(1-5)
Only TAER affects AS3X/SAFE react dir
OK, I understand that.

Than states:
If changes, RX 'Relearn Servo'
This is the part I cant figure out, how do I relearn servo?

This is the first time I ran into this issue after using your great DSM Tools. Not sure why it wont reverse and maybe relearn might fix it?
Thanks.....

txkflier, 2025-09-02:
An AR636 doesn't have Forward Programming; therefore, it doesn't have "Relearn Servo Settings". With a Spektrum transmitter, you must set the reversing to Normal and do the reversing in the receiver using the Spektrum Programmer software. I don't know about the reversing in an EdgeTX transmitter.

Are you really using an AR636 or is that a typo?

blue68camaro-git, 2025-09-02:
I was just about to edit this and say my bad. It's an AR636A.  The A is programmable but has to be done through the USB cable  as ror1948 stated. I have not used Spektrum in 2 years and forgot about this. I was revving an old plane and I should just swap it out with a FrSky. I go find my Windows machine somewhere and give it a try, yes I have the cable. Thanks for your responses.

txkflier, 2025-09-02:
The early AR636A's that came in a few types of BNF planes were locked and couldn't be updated or reprogrammed. The later AR636A's, the AR636's, and the AR636B's could be reprogrammed. They still work very well, but I've slowly been replacing them with Smart AS3X/SAFE receivers that have full-range telemetry. I like being able to monitor the flight pack voltage in my EDF's in addition to the ease of programming them.

frankiearzu, 2025-09-03:
You can reprogram most of the AR636 functionality using the SAFE Model Builder tool on the PC. I only have AR636Bs that i have used this tool with.

https://www.rcgroups.com/forums/showthread.php?2899256-Modify-Settings-on-an-AR636-SAFE-Select-Receiver

closed 2025-09-03 by frankiearzu